BehaviorTree/BehaviorTree.ROS2 #3: [Docker] failing build due to emitStateChanged() not being declared in the BehaviorTree.ROS2 package?

azalutsky, 2022-11-03:
Hi all, 
I have a docker image with ros2 and Im trying to build behaviortree and groot together to start making some cool stuff. I was wondering if you guys have any idea why my build is failing: 

The Dockerfile that does the behaviortree part of the build (using ubuntu 20.04 and ros2 foxy): 
```
RUN mkdir -p $HOME/$EBOTS_ARMS_WORKSPACE/behaviortree_cpp && \
    git clone \
    https://github.com/BehaviorTree/BehaviorTree.CPP.git \
    $HOME/$EBOTS_ARMS_WORKSPACE/behaviortree_cpp && \
    cd $HOME/$EBOTS_ARMS_WORKSPACE/behaviortree_cpp && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make && \
    make install

RUN mkdir -p $HOME/$EBOTS_ARMS_WORKSPACE/behaviortree_ros && \
    git clone \
    https://github.com/BehaviorTree/BehaviorTree.ROS2.git \
    $HOME/$EBOTS_ARMS_WORKSPACE/behaviortree_ros

# I don't think the submodules are necessary if we do the rosdep install? *unsure*
RUN mkdir -p $HOME/$EBOTS_ARMS_WORKSPACE/groot && \
    git clone \
    https://github.com/BehaviorTree/Groot \
    $HOME/$EBOTS_ARMS_WORKSPACE/groot && \
    cd $HOME/$EBOTS_ARMS_WORKSPACE/groot && \
    git submodule update --init --recursive && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make

# Install dependencies of all packages
RUN cd $HOME/$EBOTS_ARMS_WORKSPACE && \
    rosdep install -r \
    --from-paths . -y
```

The error: 
```
--- stderr: behaviortree_ros2
In file included from /home/ebots/workspace/behaviortree_ros/test/test_client.cpp:1:
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp: In lambda function:
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:217:7: error: there are no arguments to ‘emitStateChanged’ that depend on a template parameter, so a declaration of ‘emitStateChanged’ must be available [-fpermissive]
  217 |       emitStateChanged();
      |       ^~~~~~~~~~~~~~~~
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:217:7: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp: In lambda function:
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:225:7: error: there are no arguments to ‘emitStateChanged’ that depend on a template parameter, so a declaration of ‘emitStateChanged’ must be available [-fpermissive]
  225 |       emitStateChanged();
      |       ^~~~~~~~~~~~~~~~
/home/ebots/workspace/behaviortree_ros/test/test_client.cpp: In function ‘int main(int, char**)’:
/home/ebots/workspace/behaviortree_ros/test/test_client.cpp:112:10: error: ‘class BT::Tree’ has no member named ‘sleep’
  112 |     tree.sleep(std::chrono::milliseconds(100));
      |          ^~~~~
In file included from /home/ebots/workspace/behaviortree_ros/test/test_client.cpp:1:
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp: In instantiation of ‘BT::NodeStatus BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]’:
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:177:14:   required from here
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:217:23: error: ‘emitStateChanged’ was not declared in this scope
  217 |       emitStateChanged();
      |       ~~~~~~~~~~~~~~~~^~
/home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:225:23: error: ‘emitStateChanged’ was not declared in this scope
  225 |       emitStateChanged();
      |       ~~~~~~~~~~~~~~~~^~
In file included from /usr/include/c++/9/future:48,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:18,
                 from /home/ebots/workspace/behaviortree_ros/include/behaviortree_ros2/bt_action_node.hpp:6,
                 from /home/ebots/workspace/behaviortree_ros/test/test_client.cpp:1:
/usr/include/c++/9/bits/std_function.h: At global scope:
/usr/include/c++/9/bits/std_function.h:523:2: error: ‘std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]::<lambda(rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep>::SharedPtr, std::shared_ptr<const behaviortree_ros2::action::Sleep_Feedback_<std::allocator<void> > >)>; _Res = void; _ArgTypes = {std::shared_ptr<rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep> >, std::shared_ptr<const behaviortree_ros2::action::Sleep_Feedback_<std::allocator<void> > >}; std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> = std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep> >, std::shared_ptr<const behaviortree_ros2::action::Sleep_Feedback_<std::allocator<void> > >)>&]’, declared using local type ‘BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]::<lambda(rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep>::SharedPtr, std::shared_ptr<const behaviortree_ros2::action::Sleep_Feedback_<std::allocator<void> > >)>’, is used but never defined [-fpermissive]
  523 |  operator=(_Functor&& __f)
      |  ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:523:2: error: ‘std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]::<lambda(const WrappedResult&)>; _Res = void; _ArgTypes = {const rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep>::WrappedResult&}; std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> = std::function<void(const rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep>::WrappedResult&)>&]’, declared using local type ‘BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]::<lambda(const WrappedResult&)>’, is used but never defined [-fpermissive]
/usr/include/c++/9/bits/std_function.h:523:2: error: ‘std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]::<lambda(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep> > >)>; _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep> > >}; std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_Functor>::type>, std::function<_Res(_ArgTypes ...)>&> = std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep> > >)>&]’, declared using local type ‘BT::RosActionNode<ActionT>::tick() [with ActionT = behaviortree_ros2::action::Sleep]::<lambda(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<behaviortree_ros2::action::Sleep> > >)>’, is used but never defined [-fpermissive]
make[2]: *** [CMakeFiles/test_client.dir/build.make:63: CMakeFiles/test_client.dir/test/test_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/test_client.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< behaviortree_ros2 [31.0s, exited with code 2]
Aborted  <<< behaviortree_cpp [1min 50s]

Summary: 5 packages finished [1min 51s]
  1 package failed: behaviortree_ros2
  6 packages aborted: behaviortree_cpp control_msgs controller_manager_msgs groot industrial_msgs rviz_visual_tools
  2 packages had stderr output: behaviortree_cpp behaviortree_ros2
  10 packages not processed
The command '/bin/sh -c cd $HOME/$EBOTS_ARMS_WORKSPACE &&     . /opt/ros/$ROS_DISTRO/setup.sh &&     colcon build --allow-overriding control_msgs      controller_interface hardware_interface     joint_state_publisher joint_state_publisher_gui     controller_manager ros2_control_test_assets --cmake-args -DCMAKE_BUILD_TYPE=Release' returned a non-zero code: 2

```

azalutsky, 2022-11-03:
Ah I see, I needed version v3.8. instructions should be a little more clear that git clone -b v3.8 is needed for this to work at the moment. N

facontidavide, 2023-08-24:
Only support 4.X.
We don't support 3.8 anymore

closed 2023-08-24 by facontidavide