BehaviorTree/BehaviorTree.ROS2 #5: rosdep cannot locate behaviortree_cpp dependecy.

hanshans2002, 2023-03-09:
Hello!
I am struggeling to integrate behaviortree_cpp version 4.0.1 into my ROS2 package. I cannot even pass the first step, which would be including the dependencies in CMakeLists.txt and package.xml.

With "rosdep install -i --from-path src --rosdistro humble -y" I am always getting the error:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
behavior_tree_example: Cannot locate rosdep definition for [behaviortree_cpp]
`

If I add the behaviortree_cpp_v3 dependencies, there is no problem at all. 
Is it somehow not integrated, i.e. do I have to integrate it manually?

Thanks in advance,
best,
hans

facontidavide, 2023-03-09:
check branch V4

https://github.com/BehaviorTree/BehaviorTree.ROS2/tree/v4

facontidavide, 2023-03-09:
also, rosdep will fail because I haven't released as Debian packages version 4, yet.

I am waiting for version 4.1 to be stable

closed 2023-03-09 by facontidavide

hanshans2002, 2023-03-09:
Alright, thank you very much for that fast response!
Best,
Hans