BehaviorTree/BehaviorTree.ROS2 #10: foxy support

mnumanuyar, 2023-04-11:
in README.md it says this is supported after galactic. I assume there is no planned support for foxy, I am just curious what is the main reason for this

mnumanuyar, 2023-04-11:
fyi this docker builds:
```
# syntax=docker/dockerfile:1

# git clone --recursive --shallow-submodules --depth 1 git@github.com:BehaviorTree/BehaviorTree.CPP.git
# git clone --recursive --shallow-submodules --depth 1 -b humble git@github.com:BehaviorTree/BehaviorTree.ROS2.git

FROM osrf/ros:foxy-desktop
COPY BehaviorTree.CPP /BehaviorTree.CPP
WORKDIR /BehaviorTree.CPP
RUN apt-get update &&\
    apt-get -y --quiet --no-install-recommends install \
    libzmq3-dev && \
    apt-get -y autoremove && \
    apt-get clean autoclean && \
    rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/* && \
    rosdep install -r --from-paths . -y #&& \
    mkdir build && \
    cd build && \
    cmake .. && \
    make && \
    make install
COPY BehaviorTree.ROS2 /BehaviorTree.ROS2
WORKDIR /BehaviorTree.ROS2
RUN apt-get update &&\
    apt-get -y --quiet --no-install-recommends install \
    libzmq3-dev && \
    apt-get -y autoremove && \
    apt-get clean autoclean && \
    rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/* && \
    rosdep install -r --from-paths . -y #&& \
    mkdir build && \
    cd build && \
    cmake .. && \
    make && \
    make install

# generate image with: docker build -f bt_cpp.Dockerfile -t bt .
```

mnumanuyar, 2023-04-12:
there was a typo at `rosdep install -r --from-paths . -y #&& \` where # caused make steps to be skipped. so above docker does not build.

Still I am curious about the reason of foxy not being supported.

trying colcon build give errors related to `rosidl_get_typesupport_target` so maybe it is about that, I will look to it but any info is also appriciated

facontidavide, 2023-04-12:
The main reason is that I am using  callback groups and Foxy doesn't have some functionalities implemented.

I guess it COULD be possible to have a working Foxy branch, with some work

mnumanuyar, 2023-04-13:
I will work on docker for now to try it out, if everything (projects i work on) works I will probably migrate. if not, i might try things and if so, report them here.

MatteoAlbi, 2023-05-25:
I'm currently trying to readapt this library to ros2foxy, and personally I find it quite straigthforward. About this: I'd like to ask few questions:
- In the library wrappers, the executor used is always single-threaded, with one mutually-exclusive callback group, thus why it is implemented using ros galactic? It seems to me that a foxy executor would work in this case.
- In `bt_service_node.hpp`, in the tick method, why at start `spin_some` is called? Maybe to receive messages from possible subscribers? But why at the end `RUNNING` is returned? In which cases is `if (status() == BT::NodeStatus::RUNNING)` skipped?

jayv, 2023-06-22:
With both galactic and foxy EOL, have folks explored running on ROS2 Humble (LTS) and if not possible, where is this on the roadmap @facontidavide ?

mnumanuyar, 2023-06-23:
@jayv If i understand correctly, you are asking for humble support. in [readme.md ](https://github.com/BehaviorTree/BehaviorTree.ROS2) it says bt is compatible only with galactic or older, which includes humble. Also I tried with humble (in a docker) and everything works

facontidavide, 2023-06-23:
yes, this repo will support ONLY Humble or older.

I m not maintaining any End Of Life version.

closed 2023-06-23 by facontidavide

mnumanuyar, 2023-06-24:
@facontidavide  I think a clarification is needed (or it might be my bad english)

In my previous  comment when i say typo I meant usage of "older" is wrong and "newer" should be used instead.

when we say humble and older, i understand humble and all previous ros2 nodes (humble, galactic, foxy ...) and no support for newer version (iron, rolling) which contradicts with what you say about end of life versions.

I think in readme it should either say "Humble or newer", which would include Humble, Iron, rolling and updated as EoL moves

or 

It should say "only supports Humble or older except End of life versions" which only includes Humble.

Assuming Humble or newer is supported, since foxy is now in EoL, something like "all non-EoL ROS 2 versions are supported" (or whatever the correct term for non EoL) can be used in readme.

facontidavide, 2023-06-26:
It was my fault. Of course it is "newer"

marip8, 2024-02-01:
FWIW, I'm supporting an application with this repo in Foxy and Humble, and I have a [branch](https://github.com/marip8/BehaviorTree.ROS2/tree/update/foxy-rebase) based on [d4da1c](https://github.com/BehaviorTree/BehaviorTree.ROS2/commit/d4da1cfb3069e813409124376be5902ff1054dfc) (`humble` branch head as of today) that works in both distros with minimal changes