BehaviorTree/BehaviorTree.ROS2 #11: Getting rcutils_set_error_state for the given examples

ErickKramer, 2023-04-14:
Hi,

First of all, thank you for providing such a wrapper and examples on how to use it.

I was testing the `sleep_client` and `sleep_server` example and I am getting the following error

Error for the `sleep_client`

```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Service type support not from this implementation. Got:
    Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at ./src/type_support_dispatch.hpp:111
    Could not load library libbehaviortree_ros2__rosidl_typesupport_introspection_cpp.so: dlopen error: libbehaviortree_ros2__rosidl_typesupport_introspection_cpp.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/type_support_dispatch.hpp:76
while fetching it, at ./src/rmw_node.cpp:4503'

with this new error message:

  'type_support is null, at ./src/rmw_node.cpp:4573'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not initialize rcl action client: type_support is null, at ./src/rmw_node.cpp:4573, at ./src/rcl/client.c:113
[1]    149016 IOT instruction (core dumped)  ./install/behaviortree_ros2/bin/sleep_client
```

Error for the `sleep_server`
```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Service type support not from this implementation. Got:
    Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at ./src/type_support_dispatch.hpp:111
    Could not load library libbehaviortree_ros2__rosidl_typesupport_introspection_cpp.so: dlopen error: libbehaviortree_ros2__rosidl_typesupport_introspection_cpp.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/type_support_dispatch.hpp:76
while fetching it, at ./src/rmw_node.cpp:4503'

with this new error message:

  'type_support is null, at ./src/rmw_node.cpp:4573'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  type_support is null, at ./src/rmw_node.cpp:4573, at ./src/rcl/service.c:124
[1]    149048 IOT instruction (core dumped)  ./install/behaviortree_ros2/bin/sleep_server
```

I tested by building the package using ROS Humble and tested the executables by calling them directly from the build directory as `./build/behaviortree_ros2/sleep_server`

Am I missing something to correctly build/use this package?

nerovalerius, 2023-08-09:
thank you @ErickKramer, you saved me a lot of time.

facontidavide, 2023-08-24:
fixed

closed 2023-08-24 by facontidavide