BehaviorTree/BehaviorTree.ROS2 #25: goal rejected excpetion in bt_action_node

fmros, 2023-08-22:
Currently, the bt_action_node throws an exception when the action server rejects the goal, see [here](https://github.com/BehaviorTree/BehaviorTree.ROS2/blob/11f871c60d52c3bfedb2184828045263fa03e40e/include/behaviortree_ros2/bt_action_node.hpp#L366). 

I encountered some use cases where a variable is generated from a node, written through an output to the blackboard, passed as an input to an action node and then used in the goal request. This runtime generation of the goal request can lead to situations where the action server rejects it. A goal rejection could however be resolved in the behavior tree through a fallback etc. if the bt_action_node would not throw but return NodeStatus::Failure instead.

Let me know if you agree so i can file a PR. thanks

facontidavide, 2023-08-24:
can't address this without a reproducible example

EGAlberts, 2023-11-15:
I am facing the same issue. In my case the example is with Nav2, if I have one node which sends a initial position to Nav2, and then the next node sends a goal position to travel to (using the NavigateToPose action), the goal is rejected because Nav2 is still busy processing my publishing of an initial position. Giving you code to reproduce this accurately might be tricky, but I would also like to know the rationale for throwing a runtime_error at all rather than NodeFailure, since the former crashes the entire BT, and the latter allows the developer to choose how to handle the situation.

marj3220, 2023-11-27:
We are facing the same issue with a custom action. In our case, the action server is responsible to dock and undock the robot to it's docking station. If the docking station is not visible by our camera then the action server will reject the action.

The desired behavior would be to catch the action rejection in the BT and handle it accordingly. Therefore, the bt_action_node must not throw because the whole bt crashes. 

I filed a very simple PR to address this issue, please have a look and give me your feedback on this :) 
https://github.com/BehaviorTree/BehaviorTree.ROS2/pull/39

closed 2023-11-27 by facontidavide