BehaviorTree/BehaviorTree.ROS2 #26: Segmentation Fault when ticking the tree

RdDevel, 2023-08-30:
Hi everyone. 
I'm having some issues with my first behavior tree implementation.

I created a simple action client, that inherits from the RosActionNode class, in order to send target poses to a 6DOF manipulator but as soon as this client is ticked, I get a segmentation fault.

The main node, i.e. the one that "builds" and ticks the behavior tree, is the following: 
`#include "segmentation_bt_client.hpp"
#include "move_bt_action.hpp"

#include "behaviortree_cpp/bt_factory.h"
#include "behaviortree_cpp/loggers/bt_cout_logger.h"
#include "behaviortree_ros2/plugins.hpp"

using namespace BT;

int main(int argc, char** argv)
{

    const std::string bt_xml_path = "/home/diruscio/ros2_proj/src/parcel_pkg/config/bt_xml.xml";
    rclcpp::init(argc,argv);

    auto node = std::make_shared<rclcpp::Node>("bt_main");
    node->declare_parameter("bt_xml", rclcpp::ParameterValue(bt_xml_path));
    std::string bt_xml;
    node->get_parameter("bt_xml", bt_xml);

    
    RosNodeParams params_action; 
    params_action.nh = node;
    params_action.default_port_value="move_action_server";

    RosNodeParams params_service;
    params_service.nh = node;
    params_service.default_port_value = "cloud_segmentation_server";

    BehaviorTreeFactory factory;
    factory.registerNodeType<MoveToTargetAction>("MoveToTarget", params_action);    
    factory.registerNodeType<SegmentationClient>("SegmentationClient", params_service);
    
    auto tree = factory.createTreeFromFile(bt_xml);
    tree.tickWhileRunning();
    std::this_thread::sleep_for(std::chrono::milliseconds(10000));

    return 0;
}`

And the XML file is: 
`<root BTCPP_format="4" >
    <BehaviorTree ID="MainTree">
        <Sequence name="root_sequence">
            <MoveToTarget     move_type="home" target_frame="{target_frame}" />
            <SegmentationClient target_frame="{target_frame}"/>
            <Repeat num_cycles="3">
                <Sequence>
                    <MoveToTarget     move_type="pick"    target_frame="{target_frame}"/>
                    <MoveToTarget     move_type="away"    target_frame="{target_frame}"/>
                    <MoveToTarget     move_type="release"    target_frame="{target_frame}"/>            
                    <SegmentationClient target_frame="{target_frame}"/>
                </Sequence>
            </Repeat>
        </Sequence>
    </BehaviorTree>
</root>`

When running the main node, I get the following output:
[INFO] [1693418921.313392475] [move_client_logger]: Reading fixed poses from file: /home/diruscio/ros2_proj/src/parcel_pkg/config/fixed_positions.yaml 
[INFO] [1693418921.313670412] [move_client_logger]: Fixed poses read correctly
[INFO] [1693418921.848795157] [segmentation_client_logger]: Segmentation client initialized
[INFO] [1693418921.851096244] [move_client_logger]: Reading fixed poses from file: /home/diruscio/ros2_proj/src/parcel_pkg/config/fixed_positions.yaml 
[INFO] [1693418921.851268127] [move_client_logger]: Fixed poses read correctly
[INFO] [1693418921.872170030] [move_client_logger]: Reading fixed poses from file: /home/diruscio/ros2_proj/src/parcel_pkg/config/fixed_positions.yaml 
[INFO] [1693418921.872526930] [move_client_logger]: Fixed poses read correctly
[INFO] [1693418921.883179841] [move_client_logger]: Reading fixed poses from file: /home/diruscio/ros2_proj/src/parcel_pkg/config/fixed_positions.yaml 
[INFO] [1693418921.884537858] [move_client_logger]: Fixed poses read correctly
[INFO] [1693418921.888328717] [segmentation_client_logger]: Segmentation client initialized
[INFO] [1693418921.888423013] [move_client_logger]: Move to home position.
[INFO] [1693418921.888430763] [move_client_logger]: Move type: PTP
[INFO] [1693418921.888433997] [move_client_logger]: Speed: 1.000
[INFO] [1693418921.888438358] [move_client_logger]: Target x: -0.027
[INFO] [1693418921.888441878] [move_client_logger]: Target y: 0.744
[INFO] [1693418921.888444680] [move_client_logger]: Target z: 0.659
[INFO] [1693418921.888447384] [move_client_logger]: Target qx: -0.707
[INFO] [1693418921.888450183] [move_client_logger]: Target qy: -0.674
[INFO] [1693418921.888452922] [move_client_logger]: Target qz: 0.177
[INFO] [1693418921.888455738] [move_client_logger]: Target qw: -0.120
[ros2run]: Segmentation fault

The target pose is printed before the "setGoal" function returns "true".  
The action server does not receive any request. 
I sure I'm facing a trivial bug, but I have not been able to find it. 

Additional information:
OS: Ubuntu 22.04 
ROS Version: ROS 2 Humble
BehaviorTree.ROS2 version: 0.2.0

facontidavide, 2023-08-31:
Hi

without a backtrace, it is hard to define if this problem happens in the library or your code.

RdDevel, 2023-09-04:
Hi Davide, after a some test I realised that the problem was caused by these two lines of the "MoveToTargetAction" : 
`            if(action_client_ptr_->wait_for_action_server(std::chrono::seconds(timeout)))
            {
                RCLCPP_ERROR(move_client_logger, "Action server not available after %d seconds", timeout);
            }
`
These lines were left over from an earlier implementation and I think they are not needed with the actual BehaviorTree.ROS2 implementation. Thus, I'm closing the issue. 
Thank you for the support.

closed 2023-09-04 by RdDevel