BehaviorTree/BehaviorTree.ROS2 #41: Tree calling other trees design question

li9i, 2023-12-14:
I am facing a situation of the following form: There exist one central pc, one robot of type A, and one robot of type B. The pc is designed to be running a BT. That BT needs, on occasion, among other things, to make robot A and robot B perform certain actions, e.g. check self diagnostics, initialise navigation goals, make them navigate somewhere, etc. 

At this point I have only bundled up all actions that pertain to each robot type to separate BTs, i.e. there are 2 BTs: `BT_A` and `BT_B`. My question is: how would it be best in your opinion to implement `BT_pc` so that it calls functionalities (tree nodes) of `BT_A` and `BT_B`? Is something like this even feasible?

I understand that the answer to this question is part ROS-2-implementation-specific and part Behavior-tree-related. If you believe that it is best suited for the main BehaviorTree repository, or elsewhere, please point it out to me. Thank you.

marj3220, 2024-02-12:
I would create action and service nodes directly in ROS, not using the BehaviorTree system, that would accomplish the functionalities when called by the BT_pc. Your BehaviorTree nodes in BT_pc would call these actions and/or services. These BT nodes would be similar to [sleep_action.hpp](https://github.com/BehaviorTree/BehaviorTree.ROS2/blob/humble/btcpp_ros2_samples/src/sleep_action.hpp). The way BT_pc and your robots would communicate is through [ROS2 Multiple Machines](https://roboticsbackend.com/ros2-multiple-machines-including-raspberry-pi/).

li9i, 2024-02-13:
Thank you very much for your reply. In the end I finally designed this in the way you describe it. `BT_pc` hosts an action client and calls the action servers at `A` and `B` with variables pertaining to which tree should be loaded. I did not close the issue in case there was a better idea. Feel free to open it again in that case.

closed 2024-02-13 by li9i