BehaviorTree/BehaviorTree.ROS2 #48: onFeedback for bt_action_node is not receiving feedback

marj3220, 2024-02-16:
When creating a custom RosActionNode and overriding the onFeedback method, the method is never called. Here is an example:
The .hpp file
 ```
#pragma once

#include <behaviortree_ros2/bt_action_node.hpp>
#include <iostream>

#include "behaviortree_cpp/action_node.h"
#include "dictator_behavior_tree/pose_to_string.hpp"
#include "geometry_msgs/msg/pose.hpp"
#include "nav2_msgs/action/navigate_to_pose.hpp"
#include "rclcpp/rclcpp.hpp"
#include "rclcpp_action/rclcpp_action.hpp"

using Nav2 = nav2_msgs::action::NavigateToPose;

namespace BT
{
class NavigateToPose : public RosActionNode<Nav2>
{
  public:
    NavigateToPose(const std::string &name, const NodeConfig &conf, const RosNodeParams &params)
        : RosActionNode<Nav2>(name, conf, params)
    {
    }

    static PortsList providedPorts()
    {
        return providedBasicPorts({InputPort<Position2D>("goal_pose")});
    }

    bool setGoal(RosActionNode::Goal &goal) override;

    NodeStatus onResultReceived(const WrappedResult &wr) override;

    NodeStatus onFailure(ActionNodeErrorCode error) override;

    NodeStatus onFeedback(const std::shared_ptr<const Feedback> feedback) override;
};
} // namespace BT
```
the implemented method:

```
NodeStatus NavigateToPose::onFeedback(const std::shared_ptr<const Feedback> feedback)
{
    std::cout << "FEEDBACK" << std::endl;
    std::stringstream ss;
    ss << "Estimated remaining time: " << feedback->estimated_time_remaining.sec << " ";
    std::cout << ss.str() << std::endl;
    return NodeStatus::RUNNING;
}
```

With this example, I never get an output to the terminal for the received feedback. I also validated that the action does provide a feedback by listening to the action feedback topic `ros2 topic echo /navigate_to_pose/_action/feedback`

facontidavide, 2024-02-16:
Thanks for reporting, I will have a look

Rizano1, 2024-03-10:
i have same issue, I've done echo topic on feedback action and the results come out, but onFeedback doesn't get called

Perdixky, 2024-03-13:
我的程序同样无法接收到反馈，事实上，它完全没有进入onFeedback函数
`#include "rclcpp/rclcpp.hpp"
#include "rclcpp_action/rclcpp_action.hpp"
#include "action_tutorials_interfaces/action/fibonacci.hpp"

class server final : public rclcpp::Node
{
public:
    server() : rclcpp::Node("fibonacci")
    {
        this->server_ = rclcpp_action::create_server<action_tutorials_interfaces::action::Fibonacci>(
            this,
            "fibonacci",
            [this](const rclcpp_action::GoalUUID&,
                   std::shared_ptr<const action_tutorials_interfaces::action::Fibonacci::Goal> goal) -> rclcpp_action::GoalResponse{
                        return this->handle_goal(goal);
                    },
            [this](const std::shared_ptr<rclcpp_action::ServerGoalHandle<action_tutorials_interfaces::action::Fibonacci>>) -> rclcpp_action::CancelResponse
                    {
                        return this->handle_cancel();
                    },
            [this](const std::shared_ptr<rclcpp_action::ServerGoalHandle<action_tutorials_interfaces::action::Fibonacci>> goal_handle){
                        this->handle_accept(goal_handle);
                    }
        );
    }
private:
    rclcpp_action::Server<action_tutorials_interfaces::action::Fibonacci>::SharedPtr server_;
    int times;
    std::vector<int> result;
    

    rclcpp_action::GoalResponse handle_goal(std::shared_ptr<const action_tutorials_interfaces::action::Fibonacci::Goal> Goal)
    {
        if(Goal->order <= 0)
        {
            RCLCPP_ERROR(this->get_logger(), "不能是非正数！");
            return rclcpp_action::GoalResponse::REJECT;
        }
        times = Goal->order;
        return rclcpp_action::GoalResponse::ACCEPT_AND_EXECUTE;
    }

    rclcpp_action::CancelResponse handle_cancel()
    {
        RCLCPP_INFO(this->get_logger(), "Received request to cancel goal");
        return rclcpp_action::CancelResponse::ACCEPT;
    }

    auto calculate_fibonacci(const std::shared_ptr<rclcpp_action::ServerGoalHandle<action_tutorials_interfaces::action::Fibonacci>> goal_handle) -> void
    {
        std::shared_ptr<action_tutorials_interfaces::action::Fibonacci_Feedback> feedback = std::make_shared<action_tutorials_interfaces::action::Fibonacci_Feedback>();
        feedback->partial_sequence.emplace_back(0);
        feedback->partial_sequence.emplace_back(1);
        goal_handle->publish_feedback(feedback);
        std::shared_ptr<action_tutorials_interfaces::action::Fibonacci_Result> result = std::make_shared<action_tutorials_interfaces::action::Fibonacci_Result>();

        for(int t{ 0 }; t < times; t++)
        {
            if(goal_handle->is_canceling())
            {
                result->sequence = feedback->partial_sequence;
                goal_handle->canceled(result);
            }
            feedback->partial_sequence.emplace_back(feedback->partial_sequence.back() + feedback->partial_sequence[feedback->partial_sequence.size() - 2]);
            goal_handle->publish_feedback(feedback);
            std::this_thread::sleep_for(std::chrono::milliseconds(5));  //publish_feedback非堵塞，必须停止当前进程
        }
        result->sequence = feedback->partial_sequence;
        goal_handle->succeed(result);
        RCLCPP_INFO(this->get_logger(), "成功！");
    }

    void handle_accept(const std::shared_ptr<rclcpp_action::ServerGoalHandle<action_tutorials_interfaces::action::Fibonacci>> goal_handle)
    {
        times = goal_handle->get_goal()->order;
        std::thread(&server::calculate_fibonacci, this, goal_handle).detach();
    }
};

auto main(int argc, char** argv) -> int
{
    rclcpp::init(argc, argv);
    std::shared_ptr<server> ser = std::make_shared<server>();
    rclcpp::spin(ser);
    rclcpp::shutdown();
    return 0;
}`

sa32953, 2024-03-14:
I have the same issue, onFeedback doesn't get called and Tree doesnt return RUNNING As Status. Did someone find a solution for this?

marj3220, 2024-05-23:
I don't know what changed, but I now receive feedback

OriolRuizCelada, 2025-01-10:
I am having the same issue with sending a FollowJointTrajectory ros control action. The action is received, accepted and reached by the action server, but the  onFeedback and onResultReceived functions aren't triggered. Has any progress been made with this issue?