BehaviorTree/BehaviorTree.ROS2 #57: Cancelation of unkown goal not detectable

robin-mueller, 2024-04-20:
Hi, I wanted to note that **it can be desirable to throw** `rclcpp_action::exceptions::UnknownGoalHandleError` or at least any kind of exception on `cancelGoal()` when the goal response has not arrived yet. In fact, I am relying on such a case in my application, because I want to make sure that the goal is canceled either way, so I wait until it has arrived and proceed with the cancellation. After merging this pull request, there is no way for me to detect if haltTree() ran into a problem or not. Before I did like below:

```cpp
try {
  tree.haltTree();
} catch (const rclcpp_action::exceptions::UnknownGoalHandleError& e) {
    // If tree is halted directly after an action started, the goal response might not have been
    // received yet. In this case we assume the goal is about to arrive and asynchronously retry to HALT
}
```

I have a seperate ROS2 node that tries to halt the tree and doesn't know about when a RosActionNode published goal requests, so it is mandatory for this case to be detectable during halting the tree.

I am falling back to a previous commit, until this issue has been resolved.

_Originally posted by @robin-mueller in https://github.com/BehaviorTree/BehaviorTree.ROS2/issues/53#issuecomment-2067806301_

vicmassy, 2024-05-02:
probably here the approach would be to try to cancel the goal if it was sent but not accepted yet. I also share what @facontidavide said [here](https://github.com/BehaviorTree/BehaviorTree.ROS2/pull/53#issuecomment-2074990209) about not throwing errors which is quite undesirable normally. 

This is the current code: 
```cpp
if (!goal_handle_)
  {
    RCLCPP_WARN( node_->get_logger(), "cancelGoal called on an empty goal_handle");
    return;
  }
``` 

and it could be changed to something like this:
```cpp
if (!goal_handle_)
  {
   if (future_goal_handle_) {
     // Here the discussion is if we should block or put a timer for the waiting
     auto ret = callback_group_executor_.spin_until_future_complete(future_goal_handle_, timeout);
     if (ret != rclcpp::FutureReturnCode::SUCCESS) {
       // In that case the goal was not accepted or timed out so probably we should do nothing.
      }
    } else {
       RCLCPP_WARN( node_->get_logger(), "cancelGoal called on an empty goal_handle");
       return;
   }
  }
```

facontidavide, 2024-05-02:
I will adapt this change for the time being and decide later if there is any better alternative.

closed 2024-05-07 by facontidavide

robin-mueller, 2024-05-20:
I am sorry to come back on this, but it turns out that the code now still doesn't work correctly: 

if `goal_handle_` has been set due to a previous incoming goal response, the variable is never reset to `nullptr`, so if the next goal is requested and the original issue occurs, `goal_handle_` evaluates as true and yet another `UnknownGoalError` is thrown, although due to a different reason.

I suggest to do `goal_handle_ = nullptr` when the goal terminates, so for example inside its result callback.