BehaviorTree/BehaviorTree.ROS2 #74: sample_bt_executor.yaml contains missing plugin path

mrjogo, 2024-05-15:
This directory appears not to exist:

https://github.com/BehaviorTree/BehaviorTree.ROS2/blob/bb5535e47921a898fd60a7431300e9ca19ec3533/btcpp_ros2_samples/config/sample_bt_executor.yaml#L9

With BehaviorTree.CPP installed via`sudo apt install ros-humble-behaviortree-cpp` on Ubuntu 22.04.

Running the launchfile results in the following output, and I couldn't find any reference to `bt_plugins` in any BehaviorTree.CPP CMakeLists.txt:

```
$ ros2 launch btcpp_ros2_samples sample_bt_executor.launch.xml
[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2024-05-15-11-04-28-365715-ros-desktop-72895
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sample_bt_executor-1]: process started with pid [72896]
[INFO] [sleep_server-2]: process started with pid [72898]
[sample_bt_executor-1] [INFO] [1715796268.500210508] [bt_action_server]: Starting Action Server: behavior_server
[sample_bt_executor-1] terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
[sample_bt_executor-1]   what():  filesystem error: directory iterator cannot open directory: No such file or directory [/opt/ros/humble/share/behaviortree_cpp/bt_plugins]
[ERROR] [sample_bt_executor-1]: process has died [pid 72896, exit code -6, cmd '/home/ros/prebuilt_ws/install/btcpp_ros2_samples/lib/btcpp_ros2_samples/sample_bt_executor --ros-args --params-file /home/ros/prebuilt_ws/install/btcpp_ros2_samples/share/btcpp_ros2_samples/config/sample_bt_executor.yaml'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[sleep_server-2] [INFO] [1715796270.557774044] [rclcpp]: signal_handler(signum=2)
[INFO] [sleep_server-2]: process has finished cleanly [pid 72898]
```

PiusLim373, 2024-06-02:
Got the same problem. I solved by pulling the BehaviourTree.CPP repo into my workspace and the fix from: 
https://github.com/BehaviorTree/BehaviorTree.CPP/pull/804

Rebuild and I am able to run.

aboseley, 2024-06-13:
I hit this problem too.   Thanks @PiusLim373 for the fix