BehaviorTree/BehaviorTree.ROS2 #75: #57 works only reliably for the very first goal

robin-mueller, 2024-05-20:
As i commented in the [original issue](https://github.com/BehaviorTree/BehaviorTree.ROS2/issues/57#issuecomment-2119485893_), `cancelGoal()` still doesn't work correctly for pre-response action cancellation. The issue now is, that `goal_handle_` is not reset after a previous goal terminates, so misleadingly the function assumes for any subsequent goals that the corresponding response has already arrived.

We either have to reset `goal_handle_` accordingly, or refer to the currently relevant goal uuid if we want to verify that the response has arrived.

robin-mueller, 2024-07-10:
Is there any update on this topic?

fafux, 2025-09-10:
I had the same problem when canceling goals multiple times (working only the very first time) getting as result `rclcpp_action::UnkownGoalError`. Adding `goal_handle_ = nullptr;` at this [line](https://github.com/BehaviorTree/BehaviorTree.ROS2/blob/bb5535e47921a898fd60a7431300e9ca19ec3533/behaviortree_ros2/include/behaviortree_ros2/bt_action_node.hpp#L388) fixes.

flbeck, 2025-09-22:
I am currently experiencing a similar issue mentioned in #110 and the fix provided by @fafux seems to fix this too.