BehaviorTree/BehaviorTree.ROS2 #82: colcon build error

pipinet, 2024-07-02:
i import  [BehaviorTree.ROS2](https://github.com/BehaviorTree/BehaviorTree.ROS2) as submodule.
directory structure is 
```bash
- xxxx_ws
   - dependencies
        - BehaviorTree.CPP
        - BehaviorTree.ROS2
   - src
       - module1
       - module2
    colcon.pkg 
   pixi.toml
```
colcon.pkg 
```json
{
    "build": {
        "base-paths": ["src", "dependencies/BehaviorTree.CPP", "dependencies/BehaviorTree.ROS2"],
        "packages-skip": ["btcpp_ros2_samples"]
    }
}
````

use flow command will got error.
```bash
pixi shell

colcon build --symlink-install --cmake-args -DPython3_FIND_VIRTUALENV=ONLY
```

```log
Starting >>> behaviortree_ros2
-- Found ament_cmake: 1.3.7 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/ament_cmake/cmake)
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rclcpp: 16.0.8 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rmw_implementation_cmake/cmake)
-- Found rmw_cyclonedds_cpp: 1.3.4 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rmw_cyclonedds_cpp/cmake)
-- Using RMW implementation 'rmw_cyclonedds_cpp' as default
-- Found rclcpp_action: 16.0.8 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rclcpp_action/cmake)
-- Found behaviortree_cpp: 4.6.0 (/Users/pipinet/workspace/automation/install/behaviortree_cpp/share/behaviortree_cpp/cmake)
-- Found btcpp_ros2_interfaces: 0.2.0 (/Users/pipinet/workspace/automation/install/btcpp_ros2_interfaces/share/btcpp_ros2_interfaces/cmake)
-- Found generate_parameter_library: 0.3.7 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/generate_parameter_library/cmake)
-- Found parameter_traits: 0.3.7 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/parameter_traits/cmake)
-- Found rclcpp_lifecycle: 16.0.8 (/Users/pipinet/workspace/automation/.pixi/envs/default/share/rclcpp_lifecycle/cmake)
-- Configuring done (2.2s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/pipinet/workspace/automation/build/behaviortree_ros2
[ 20%] Running `/Users/pipinet/workspace/automation/.pixi/envs/default/bin/generate_parameter_library_cpp /Users/pipinet/workspace/automation/build/behaviortree_ros2/bt_executor_parameters/include//bt_executor_parameters.hpp /Users/pipinet/workspace/automation/dependencies/BehaviorTree.ROS2/behaviortree_ros2/src/bt_executor_parameters.yaml `
[ 20%] Built target bt_executor_parameters
[ 40%] Building CXX object CMakeFiles/behaviortree_ros2.dir/src/bt_ros2.cpp.o
[ 60%] Building CXX object CMakeFiles/behaviortree_ros2.dir/src/bt_utils.cpp.o
[ 80%] Building CXX object CMakeFiles/behaviortree_ros2.dir/src/tree_execution_server.cpp.o
/Users/pipinet/workspace/automation/dependencies/BehaviorTree.ROS2/behaviortree_ros2/src/tree_execution_server.cpp:206:28: warning: format string is not a string literal (potentially insecure) [-Wformat-security]
      RCLCPP_WARN(kLogger, action_result->return_message.c_str());
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rclcpp/rclcpp/logging.hpp:972:7: note: expanded from macro 'RCLCPP_WARN'
      __VA_ARGS__); \
      ^~~~~~~~~~~
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rcutils/rcutils/logging_macros.h:750:5: note: expanded from macro 'RCUTILS_LOG_WARN_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rcutils/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/Users/pipinet/workspace/automation/dependencies/BehaviorTree.ROS2/behaviortree_ros2/src/tree_execution_server.cpp:206:28: note: treat the string as an argument to avoid this
[... 85 lines cut ...]
    __VA_ARGS__)
    ^~~~~~~~~~~
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rcutils/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/Users/pipinet/workspace/automation/dependencies/BehaviorTree.ROS2/behaviortree_ros2/src/tree_execution_server.cpp:274:27: note: treat the string as an argument to avoid this
    RCLCPP_ERROR(kLogger, action_result->return_message.c_str());
                          ^
                          "%s",
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rclcpp/rclcpp/logging.hpp:1407:7: note: expanded from macro 'RCLCPP_ERROR'
      __VA_ARGS__); \
      ^
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rcutils/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^
/Users/pipinet/workspace/automation/.pixi/envs/default/include/rcutils/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^
4 warnings and 1 error generated.
make[2]: *** [CMakeFiles/behaviortree_ros2.dir/src/tree_execution_server.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/behaviortree_ros2.dir/all] Error 2
make: *** [all] Error 2
Failed   <<< behaviortree_ros2 [13.7s, exited with code 2]
```

KaiYuanBu, 2024-07-15:
I also encountered a colcon build error but mine is about missing packages:

> Starting >>> btcpp_ros2_interfaces
> Finished <<< btcpp_ros2_interfaces [4.35s]                    
> Starting >>> behaviortree_ros2
> --- stderr: behaviortree_ros2                         
> CMake Error at CMakeLists.txt:19 (find_package):
>   By not providing "Findgenerate_parameter_library.cmake" in
>   CMAKE_MODULE_PATH this project has asked CMake to find a package
>   configuration file provided by "generate_parameter_library", but CMake did
>   not find one.
> 
>   Could not find a package configuration file provided by
>   "generate_parameter_library" with any of the following names:
> 
>     generate_parameter_libraryConfig.cmake
>     generate_parameter_library-config.cmake
> 
>   Add the installation prefix of "generate_parameter_library" to
>   CMAKE_PREFIX_PATH or set "generate_parameter_library_DIR" to a directory
>   containing one of the above files.  If "generate_parameter_library"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> ---
> Failed   <<< behaviortree_ros2 [0.92s, exited with code 1]
> 
> Summary: 1 package finished [5.51s]
>   1 package failed: behaviortree_ros2
>   1 package had stderr output: behaviortree_ros2
>   1 package not processed

tony-p, 2024-08-19:
Can you give the output of  `pixi list`?

I also build in a pixi env but don't see this issue

tony-p, 2024-08-20:
are you building on windows with MSVC?

I got some build errors on windows (not as verbose as what you get) but for me are solved by https://github.com/BehaviorTree/BehaviorTree.ROS2/pull/91